agg23/openfpga-SNES #2: Save does not stick if exiting game through the "Load SMC" menu

MysticNTN, 2022-09-16:
If I do a quick reload of a snes rom through the "Load SMC" menu the save file will not be written properly. To get the save to stick I have to shutdown the pocket entirely.

agg23, 2022-09-16:
I cannot reproduce this. A fast test is opening SMW and waiting until the world map, then switching ROMs.

Additionally, it should _definitely_ save when you quit the core. Turning off the Pocket should not be necessary.

MysticNTN, 2022-09-17:
I just tested this with SMW and it does indeed retain the save. I should have given more information in the original post, my apologies. 

The games I tested this with were
-The Legend of Zelda A Link to the Past
-Chrono Trigger

saving in both of those games respectively then switching roms resulted in a missing save when returning.

juraisa, 2022-09-17:
Also can't reproduce so far.

I loaded Link To The Past, started game, performed a Save And Quit, waited for game to restart then used Core Options to Load SMC into Mystical Ninja. Load SMC'd back into Link To The Past, save is there as expected.

MysticNTN, 2022-09-17:
[Here ](https://youtu.be/aW3itHK_CPI)is a video showing it for Link to the Past.

xportz, 2022-09-17:
This just happened to me on SMW unfortunately.  I made it past the first castle, saved, then loaded a new rom through the menu.  When I came back to SMW, my saved game was not there.

naveedster, 2022-09-17:
> 

Yeah, I guess I am used to the way that the Sd2SNES behaves.

MysticNTN, 2022-09-17:
So this also seems to lose saves of games I have not loaded. Here's what I did:

- Booted the Pocket
- Loaded Chrono Trigger and verified the save file was present.
- Restarted the Pocket
- Loaded SNES core and DKC
- Played a few levels
- Load SMC to Doom
- Played a couple minutes
- Load SMC to Chrono Trigger and the save file is missing.

agg23, 2022-09-17:
> So this also seems to lose saves of games I have not loaded

That's a huge issue if true. Not something that can be caused by the core (unless you missed something in your description) as files can only be touched via a special configuration, and save files are linked by name and path to the ROM.

MysticNTN, 2022-09-17:
I will try to make a video for it.

And yeah i was just thinking that the first hard shutdown could have caused it. I will come back with more info.

xportz, 2022-09-17:
I've played the first few levels of SMW as a test.  Here's my findings:

Attempt 1: Saving then loading a new ROM from the menu.  [Result: Lost save file.](https://youtu.be/BSdJDtGzAn0)

Attempt 2: Saving then quitting from the menu.  [Result: Successful save.](https://youtu.be/sElG6KQ9X00)

MysticNTN, 2022-09-17:
I was just able to reproduce it, I am uploading the video now.

> That's a huge issue if true. Not something that can be caused by the core (unless you missed something in your description) as files can only be touched via a special configuration, and save files are linked by name and path to the ROM.
> 
> Actually, it's quite possible that it didn't save when you closed Chrono Trigger that first time. So you loaded the save, but then overwrote it with garbage.

MysticNTN, 2022-09-17:
> That's a huge issue if true. Not something that can be caused by the core (unless you missed something in your description) as files can only be touched via a special configuration, and save files [...]

[Here is the video](https://youtu.be/spIcvcO1EfE)

xportz, 2022-09-17:
So I played some more SMW, and saved after the first ghost house.  I then quit and shut down, loaded up the game and my save was still there (showing 7 levels complete).  Next, I used "Load SMC" from the menu, jumped into another game for a few seconds and came back to SMW.  My save was deleted.  It seems "Load SMC" is incredibly dangerous in general!

agg23, 2022-09-17:
Looking through the configs, I don't think I did anything incorrectly. I will temporarily remove the option to change ROM

agg23, 2022-09-17:
@MysticNTN thank you greatly for the video. Could you try the process again, grabbing the Chrono Trigger save after each reboot, then the save at the very end? I would like to check to see if I can notice anything.

MysticNTN, 2022-09-17:
@agg23 Yes I can. One moment.

MysticNTN, 2022-09-17:
@agg23 [Here you go.](https://github.com/agg23/openfpga-SNES/files/9592482/MysticNTN-Issue.2.zip) Let me know if this helps, I also included the rom in case that reveals anything. Not sure if that is "welcomed" on GitHub 😆

agg23, 2022-09-17:
Did you follow the same pattern from before? If so, can you send me the DKC save file (the original I guess and whatever is on your Pocket now after repeating the experiment again) and any Doom saves?.

The first two saves are identical, as we would expect. The third is much larger, and contains a bunch of garbage data (looks like the default state of the memory block honestly). I'm wondering if some of the first data in that final save correlates to data in either DKC or Doom, which may shed more light on what is going on.

MysticNTN, 2022-09-17:
Yes, those saves are from the exact steps taken in the video. Do you want me to repeat those steps and send you the DKC save and whatever Doom spits out? (Im not too familiar with how the saving works in that game)

agg23, 2022-09-17:
No, you don't have to repeat it. I thought you had a pre-existing DKC save from a SuperNT or something, so I want that save and the final DKC save after things get messed up (which is what would be on your Pocket right now unless you copied over it already). And just the final Doom save (I'm assuming there was no starting Doom save).

MysticNTN, 2022-09-17:
So I just went ahead and repeated it. Sorry! But as before, the end result was missing saves so this should help you out. 
[openfpga-SNES-Issue#2-DKC-Doom.zip](https://github.com/agg23/openfpga-SNES/files/9592568/openfpga-SNES-Issue.2-DKC-Doom.zip)

agg23, 2022-09-17:
Ok, perfect. Thank you for being so helpful and repeating that over and over again. So the Doom save is actually your DKC save, which shows basically what I was thinking; that there's a firmware issue that is causing save corruption. That doesn't really explain what is happening to Chrono Trigger, but it's also just not possible from my end for Doom to get the DKC data.

I have referenced this discussion in the report to Analogue, and hopefully we have a fix soon. Thanks again for your help.

MysticNTN, 2022-09-17:
You're very welcome. If you need any more help on this issue or other please let me know. I would love to continue to help on this project.

agg23, 2022-09-20:
Going to close this for now, as I think it should be mitigated. If this still occurs, please reopen.

closed 2022-09-20 by agg23