agg23/openfpga-SNES #10: A, B, X, and Y Buttons are mapped incorrectly.

naveedster, 2022-09-17:
Tested in super Mario world. When the user presses Y while moving, they should sprint, not jump. When I press the button that should be B, the user spin jumps (spin jump should be A) but B should be regular jump. 

So A, B, X, and Y are mixed up.

Llorean, 2022-09-17:
I don't experience this at all. Super Mario World behaves as expected.

If you have MegaMan X (probably other games as well) you can go to the options menu, and re-assign buttons, and see which button it detects each of your physical buttons are detected as, as well.

naveedster, 2022-09-17:
Ok I’ll test it right now and make a video. To prove it.

naveedster, 2022-09-17:
> I don't experience this at all. Super Mario World behaves as expected.
> 
> If you have MegaMan X (probably other games as well) you can go to the options menu, and re-assign buttons, and see which button it detects each of your physical buttons are detected as, as well.

https://twitter.com/persianimm0rtal/status/1571043383241740288?s=46&t=Tx_cucYtcSeB9MYgCmga_Q

Llorean, 2022-09-17:
Have you edited any of the .json files?

agg23, 2022-09-17:
Try rebooting the Pocket. Someone reported that changing the controls in the GBA core changed controls in other cores as well. I believe a reboot fixed it.

SegaSnatcher, 2022-09-17:
I found this issue in Super Metroid, the controller menu itself is not saving the Y and B setting.  It keeps reverting back to X and A for shoot and jump.

ifighftdragons, 2022-09-17:
Can confirm that the buttons are indeed mapped correctly. This issue can be closed. Enjoy the games.

naveedster, 2022-09-17:
Nothing happens on reboot. Is there a different method to reboot?

naveedster, 2022-09-17:
I fixed it, but I still consider this am open issue. From the main menu of the analog pocket, I went to settings > pocket > systems > GBC > controls, and deselected Super GB. 

Went back and it fixed it. I still consider this an open issue because settings for other systems should not cross wires with the snes core.

agg23, 2022-09-17:
Agreed, but it's not under the core's control. It's an issue with Analogue's firmware. I've reported it to them, and since you've confirmed the controls in the SNES core are correct, I'm going to close the issue. Feel free to reopen if you think I got something wrong.

closed 2022-09-17 by agg23