agg23/openfpga-SNES #11: Trials of Mana/Seiken Densetsu 3 High Resolution Text Distorted

raydon4, 2022-09-17:
Text in Trials of Mana english rom (from Collection of Mana) appears scrunched up and almost unreadable. I assume this is due to the game using the high resolution mode (512*448) for text and the character select screen.
The game runs fine otherwise.

Llorean, 2022-09-17:
I noticed there weren't any video modes defined for the high-res mode in video.json. I'm not sure if that's necessary, but I noticed other systems that have multiple output resolutions seem to include definitions for each of them in it.

agg23, 2022-09-17:
Thanks for the report. I was looking for an example of a game that used the high resolution mode, and couldn't find a mention anywhere.

I'll look into it

raydon4, 2022-09-17:
> Thanks for the report. I was looking for an example of a game that used the high resolution mode, and couldn't find a mention anywhere.
> 
> I'll look into it

No worries. If you need another game that has the same issue then Secret of Mana also does something similar where some screens use high res text like the main menu and the status screen. I just tested it just now and it has the same issue.

closed 2022-09-20 by agg23