agg23/openfpga-SNES #17: Core does not find existing save when running same exact rom in a different folder

naveedster, 2022-09-17:
This is a quality of life issue, but many people will probably experience this.  I have my roms organized alphabetically, but also by genre. So, when I was browsing, I first ran the Zelda rom from my Genre folder, saved the game. Next time I opened the exact same rom from the alphabetical folder, and it did not find my previous save. Had to keep selecting the rome from the same subfolder, otherwise it counted as a different save.

juraisa, 2022-09-17:
This should be the expected behavior in my opinion. An individual ROM entity should have an individual save associated with it, not the expectation that you can run the ROM from any random path and always have the same save load

agg23, 2022-09-17:
I also agree this is expected behavior, but I guess it's an ok feature request for Analogue.

HeyItsLollie, 2022-09-18:
This issue is Pocket's doing, and not specific to any core. Analogue Pocket stores save files in a different folder, aptly named "Saves". The "Saves" folder mirrors the structure of your "Assets" folder and subfolders.

Basically, if you move your game file to a new folder, you also have to move your .sav file to an identically-named folder.

[image]

closed 2022-09-18 by agg23