agg23/openfpga-SNES #18: Romance of the Three Kingdoms II visual noise

kpthomas85, 2022-09-17:
For Romance of the Three Kingdoms II, there is a horizontal line across the top that does not disappear, starting in load up screen and remaining throughout gameplay. This exact rom does not display this line on Super NT or SD2snes. 

[image]

naveedster, 2022-09-17:
Not seeing this issue on a real 1chip snes, with Sony PVM monitor. 
[image]

agg23, 2022-09-17:
I'm curious if this occurs on the MiSTer core. It's clearly in the overscan area, so people are less likely to notice, but it does appear to be unique to the core. It's also possible that this game does something different with the video that I am not properly accounting for.

kpthomas85, 2022-09-17:
Yeah seems to only affect the pocket, didn’t see it on my snes jr with HDRV
cables to my crt either. I don’t have a mister so I unfortunately cannot
test that. I don’t suppose there’s a way to crop the overscan on pocket
(yet).

naveedster, 2022-09-17:
Just to clarify, I even went into my monitors service menu and adjusted the vertical center, and vertical size (stretching of the video signal) and made sure that it wasn’t hidden by the overscan like it would be in a game like Mario for NES.

naveedster, 2022-09-17:
What revision of SNES Hardware is the mister core based on? Maybe that is why?

multi-vitamin, 2022-09-18:
The same symptoms appeared in ys3 and ys4.
However, since it is a translated rom, it may not appear in the original.

Dubon77, 2022-09-18:
Not seeing this on Mister
[image]

multi-vitamin, 2022-09-18:
If you play a game with black background like final fantasy 5,
The noise line disappears.

Perhaps the noise line is masked by other games or any data.

SegaSnatcher, 2022-09-20:
What is the resolution of this game?  Maybe a scaling bug because of an odd resolution it's using?

SegaSnatcher, 2022-09-20:
Ahh,  MiSTer is reporting a resolution of 256x239 for that game where most other games are 256x224.

SegaSnatcher, 2022-09-20:
> ys3

Yep ys3 and lV also use the weird 256x239 resolution as this game.   So there is definitely an issue with that specific resolution.

kpthomas85, 2022-11-10:
Nice! Well done