agg23/openfpga-SNES #20: Live a Live Eng Patch v2.0 translation loads to black screen

fourswordslink, 2022-09-17:
Game boots to either a black screen or a black screen with a line of pixels on top.

Using fresh rom from No-Intro (has no header) + Aeon Genesis 2.0 English translation patch, I receive the above issues.
This does not happen on a non-patched JPN rom, or using the older 1.0 translation patch. Also verified the 2.0 patched rom runs in BSNES/higan.
Original:
[image]
Patched (with 2.0 translation):
[image]

agg23, 2022-09-17:
Try these instructions. It seems to have fixed several other patched ROMs with issues: https://github.com/agg23/openfpga-SNES/issues/3#issuecomment-1250088480

fourswordslink, 2022-09-17:
> Try these instructions. It seems to have fixed several other patched ROMs with issues: [#3 (comment)](https://github.com/agg23/openfpga-SNES/issues/3#issuecomment-1250088480)

That worked for me. Thank you very much

closed 2022-09-17 by fourswordslink