agg23/openfpga-SNES #3: [ Final Fantasy 5 ] Game is frozen/stuck after Title screen fade

mikey303, 2022-09-17:
Issue does not appear in the unpatched JPN rom, however after patching to english (via https://www.romhacking.net/hacks/3687/) the game will show the title, fade out, and then not proceed.  You can hear the music very softly after so it seems to be getting stuck.

ifighftdragons, 2022-09-17:
Use SNES ROM Utility to:
1. Fix Checksum
2. Fix ROM size
3. Fix Checksum again
4. Save, and try to play it.

It works on every game I've had a problem booting. Headers must also be removed, of course.

It will work, and you can then close this issue. 
Enjoy the game!

SegaSnatcher, 2022-09-17:
Thanks that worked!

agg23, 2022-09-17:
Interesting. So does patching not generally update the checksum? Weird. I do use the checksum to help validate the header.

ifighftdragons, 2022-09-17:
I've seen a lot of romhacks and translations mess up the checksums, and running a fix on them has always worked for me. Not exactly sure how or why it works, but I'm just glad it does.

mikey303, 2022-09-17:
That fixed it. TYSM!!!!

closed 2022-09-17 by mikey303

mattd121, 2022-09-20:
Worked for me! TY!!

SmiteOfHand, 2022-09-20:
Any plans to investigate this any further? I know there is the work around posted above, however the original romhacks such as this (https://www.romhacking.net/hacks/4134/) work without further modification across original SNES hardware and the MiSTer, as well as every emulator I've thrown it at (bsnes, snes9x, higan, etc..)

There appears to be an issue specific to the Pocket implementation.

agg23, 2022-09-20:
Looked into it further and just started to ignore the ROM's reported size, calculating it instead. Should be fixed in the next release.

RNekic, 2022-09-29:
> Looked into it further and just started to ignore the ROM's reported size, calculating it instead. Should be fixed in the next release.

Is this fix only for FF5?  I interpreted it to mean headers will be ignored on all ROMs but I'm finding v0.3.1 is still failing to load ROMs with headers.

agg23, 2022-09-29:
This doesn't have anything to do with SMC headered ROMs. You still need your ROMs to have no SMC header.

This just fixes a mismatch between the contents of the actual (not SMC) header and the real size of the ROM, which commonly occurs with patched games.

RNekic, 2022-09-29:
Gotcha. The changelog explanation for this fix is being misunderstood by me and others:  "Ignore ROM header and calculate ROM size manually".

I don't want to drift too far off on another topic here, but is the SMC header thing something that could be ignored?  I'm finding a decent number of my roms have headers and the SNES rom utility does not appear capable of bulk-removing them.

agg23, 2022-09-29:
It will be eventually. There was a bug in the firmware that prevented me from working around, and now it's waiting on the special magic for expansion chips to be finished