agg23/openfpga-SNES #6: Pilotwings Glitching

RetroAsylum79, 2022-09-17:
The game loads fine but when you start to actually do an event, the game starts glitching. Unplayable.

naveedster, 2022-09-17:
I believe that pilot wings uses the enhancement DSP1 chip in the cartridge which is not supported yet. So other games that will not work are Mario kart and top gear 3000, but if I recall correctly, the developer is working on that as we speak.

RetroAsylum79, 2022-09-17:
🤦🏼 I'm sorry, I feel like a dope. I read the list of supported chips and must have mixed them up in my head so when the game booted I thought it was a glitch, I apologize.

ifighftdragons, 2022-09-17:
> 🤦🏼 I'm sorry, I feel like a dope. I read the list of supported chips and must have mixed them up in my head so when the game booted I thought it was a glitch, I apologize.

You should close this issue. The creator has stated the DSP1 chip will soon be implemented.

closed 2022-09-17 by RetroAsylum79