agg23/openfpga-SNES #7: Super Famicom save files

sax-bit, 2022-09-17:
This core recognizes Super Famicom ROMs, but it doesn't seem to recognize any previous Super Famicom save files. It only recognizes my SNES save files for some reason.

Please help!

agg23, 2022-09-17:
Did you try renaming them? It only looks for files ending in `.sav`

OngoGablogian, 2022-09-17:
Also the Analogue Pocket organizes save files exactly how your ROMs are organized. If you have a ROM in a folder labeled "SNES Translations" for example, you'd need create a folder in your saves directory with the same name and place the save file in that folder. Otherwise the save file wont be read properly.

ifighftdragons, 2022-09-17:
> Also the Analogue Pocket organizes save files exactly how your ROMs are organized. If you have a ROM in a folder labeled "SNES Translations" for example, you'd need create a folder in your saves directory with the same name and place the save file in that folder. Otherwise it wont be read properly.

Doesn't the Pocket automatically create these folders for save files?

sax-bit, 2022-09-17:
Yes, all my files are in .sav format, and yes my saves are labeled AND organized exactly like my ROMs.

I figured it out though. If I don't completely quit the core when I'm ending a game, and then I try to open a game where I have a previous save, it doesn't want to read yhe save at all. It just happened to be a coincidence that this happened when I was trying to play a Super Famicom game instead of an SNES game. 

But yeah, I even noticed in another post that there is occasionally some weird stuff going on with saves with this core that requires you to quit the core completely when switching between games. I imagine my issue is related to that.

agg23, 2022-09-18:
Closing in favor of #2

closed 2022-09-18 by agg23