agg23/openfpga-SNES #9: crackling and popping audio

naveedster, 2022-09-17:
It’s very faint, but, crackling and popping audio is audible and especially noticeable when listening on headphones. Noticed on star Fox 2 intro menu, and on Super Metroid.

robhlark, 2022-09-17:
I have also noticed seemingly randomly occuring faint crackling/popping audio when listening with headphones. I've tested Super Mario All-Stars + Super Mario World (USA), The Legend of Zelda - A Link to the Past (USA) and Super Metroid (Japan, USA) and all have the same audio popping issue. I have also tested other cores (by spiritualized1997) and have not found the audio popping issue there.

Great work on this SNES core so far otherwise, thanks!

thedrdodo, 2022-09-17:
Can confirm this. It's especially noticeable with headphones on. Super Mario World and Sim City both exhibit it. Most prevalent with Sim City imo.

BooBerry, 2022-09-17:
Can confirm this too. But call me crazy if you want, but it kinda sounds like the audio is clipping when the pops happen. Like it's too loud for the speakers to handle or something.

NotSure47247323, 2022-09-18:
Getting the same issue via dock on Chrono Trigger US.

naveedster, 2022-09-18:
I believe that it’s happening in pretty much every title to some extent you can just hear it more in certain games

closed 2022-09-20 by agg23