agg23/openfpga-SNES #23: Megaman X2 Overdrive Ostrich

miavuli, 2022-09-18:
When fighting Overdrive Ostrich in Megaman X2 he will occasionally shoot a sonic slicer at you from his mouth. Normally this will move towards you but on this core it just hovers in mid air.

SegaSnatcher, 2022-09-18:
Possibly another CX4 related issue since it was not properly implemented.

miavuli, 2022-09-18:
https://user-images.githubusercontent.com/26606694/190905545-6da8930a-19b9-4379-925a-5a865d2c60ec.mp4

Here is a video of the issue

miavuli, 2022-09-18:
I also found a another problem in this stage. When riding on the bike you're supposed to shoot a bridge with it but that bridge is completely missing. I have attached a video 

https://user-images.githubusercontent.com/26606694/190906587-efe21f4a-1ea9-4516-888a-cfd621d61301.mp4

ifighftdragons, 2022-09-18:
CX4 issue. Good documentation. Please close this issue, and include this info in the other issue post. No need for two issues on the same game.

miavuli, 2022-09-18:
Adding comments to other post

closed 2022-09-18 by miavuli