agg23/openfpga-SNES #24: Megaman X2 Morph Moth Graphical Issues

miavuli, 2022-09-18:
When fighting Morph Moth he instantly becomes a yellow box and on one of his attacks he just disappears.

SegaSnatcher, 2022-09-18:
> When fighting Morph Moth he instantly becomes a yellow box and on one of his attacks he just disappears.

CX4 was not properly implemented in the current build.   This likely the issue since X2 uses that chip.

miavuli, 2022-09-18:
Here is a video of the issue.

miavuli, 2022-09-18:
I have attached a video of the issue

https://user-images.githubusercontent.com/26606694/190907679-e26b6689-b5f5-478e-a443-bdce1c5e6f21.mp4

miavuli, 2022-09-18:
Adding hovering Overdrive Ostrich issues here
Hovering sonic slicer issue 

https://user-images.githubusercontent.com/26606694/190914488-da007d8a-911b-4711-ab5a-eb26874e7edd.mp4

miavuli, 2022-09-18:
Missing bridge in Overdrive Ostrich stage

https://user-images.githubusercontent.com/26606694/190914521-bcfe1bcf-bf7d-4f47-aa4e-b424d461def7.mp4

agg23, 2022-09-20:
I'm not familiar with the games, but CX4 was not working properly in the initial release (version 0.1.0). It should now be fixed. Please reopen if these issues are not in fact fixed.

closed 2022-09-20 by agg23