agg23/openfpga-SNES #27: Half screen issue

miavuli, 2022-09-20:
After updating to the latest version I now have half a screen on every game.
[image]

agg23, 2022-09-20:
You didn't copy over all of the files. You need to copy over everything, including the `video.json`.

miavuli, 2022-09-20:
I copied over all of the files. I just unzip the main file and drop all of the folders into the root. For some reason snes_main is not there. 
[image]

agg23, 2022-09-20:
I'm not sure what you're saying. The `snes_main.rev` file is present in the zip.

miavuli, 2022-09-20:
I don't see it for some reason. I'm not sure why. When I download the zip I'm presented with these files only. 
[image]

agg23, 2022-09-20:
You aren't downloading the release, but the repo. You should be clicking on the "Release" button on the right hand side and downloading the `agg23.SNES_0.2.0_2022-09-19.zip` from https://github.com/agg23/openfpga-SNES/releases/tag/0.2.0

miavuli, 2022-09-20:
Thank you! Thats what the issue was!

closed 2022-09-20 by miavuli