agg23/openfpga-SNES #31: Mega Man X high pitched sound

agg23, 2022-09-21:
Report from https://twitter.com/MrPixelSynth:

In Mega Man X on Launch Octopus stage. There is this really weird high pitched audio noise sometimes when X is under water

He ran through the entire game and didn't notice any other issues.

miavuli, 2022-09-21:
Unless he is taking about the Utuboros serpent making a high pitch noise when you shoot it(completely normal) I can't reproduce this issue. 
[image]

agg23, 2022-09-25:
Turns out this is part of the game, it's just easier to hear on the Pocket

https://twitter.com/MrPixelSynth/status/1573696456086757382

closed 2022-09-25 by agg23