agg23/openfpga-SNES #37: Device not supported error upon NBA Jam Launch with 0.3.0 core.

FrogSeven, 2022-09-29:
With the 0.3.0 SNES Core and B6 Firmware (9/28) I get the following error message when loading several versions of NBA Jam.  (USA, USA Rev 1, and NBA Jam -Tournament Edition (USA))

"A device not supported by this program has been connected.  Please use the standard controller only"

Does not reproduce on the 0.1.0 or 0.2.0 cores on either the latest or n-1 firmware and it seems limited to the 0.3.0 core.   I haven't modified any files or settings but reset the settings anyway without any success.

agg23, 2022-09-29:
Please download 0.3.1. It should be fixed in that release

FrogSeven, 2022-09-29:
Indeed, it is fixed there.  Sorry for missing that there was a newer release.  Thanks!

closed 2022-09-29 by FrogSeven