agg23/openfpga-SNES #38: Castlevania - Dracula X crashing core

DJBabyBuster, 2022-09-29:
Trying to run Castlevania - Dracula X, I get a blank screen after load, the when trying to quit the core, it won't exit and I have to hard reboot the pocket

ifighftdragons, 2022-09-29:
Dracula X works like a charm. 
You just need to use SNES Advanced ROM Utility to either fix Checksum, ROM Size and/or remove header.

agg23, 2022-09-29:
> when trying to quit the core, it won't exit and I have to hard reboot the pocket

That's a pretty big deal. What firmware version are you running? It's stuck at a completely black screen? Do other games work?

DJBabyBuster, 2022-09-29:
The latest 0.3.1
Yeah it's the first time I've encountered a rom that crashed the core like that. 
It exhibited the same behavior 3 times, but trying again just now it won't load and I'm getting "Error in Framework File ID [10] too large"

agg23, 2022-09-29:
Deleting the save file for the game should fix that problem (but not the black screen)

agg23, 2022-10-27:
Sorry for the delay. Is this still occurring? I cannot reproduce. Please reopen if you can reproduce

closed 2022-10-27 by agg23