agg23/openfpga-SNES #40: Graphics glitches in the snes core

Urbalcloud, 2022-09-29:
noticing some glitching at the bottom of the screen on a few games in the snes core, including Super Adventure Island and Zelda: LttP. Attempting to attach pics showing the issues

[image]
[image]

italiangrandma, 2022-09-29:
I noticed the same issue in Nightmare Busters (Unlicensed).

ifighftdragons, 2022-09-29:
Tried the same games, no issues here. Their ROMs are either headered, or not Checksum/ROM size fixed.

agg23, 2022-09-29:
Are these PAL games or did you edit `video.json`? Looks like it's using the wrong `video.json` settings, which is most likely to occur with PAL

italiangrandma, 2022-09-29:
> Are these PAL games or did you edit `video.json`? Looks like it's using the wrong `video.json` settings, which is most likely to occur with PAL

I did just notice that Nightmare Busters is a PAL game after running it through the Advanced SNES Rom Utility. I also fixed the checksum and rom size. Nightmare Busters still shows the red line at the bottom, so it must be related to the game being PAL.

@ifighftdragons Did you test Nightmare Busters?

ifighftdragons, 2022-09-29:
@italiangrandma: I just tested the games pictured. PAL games are already mentioned by agg23 to not be working properly due to waiting on updates from Analogye, so it shouldn't be logged as an issue here.

italiangrandma, 2022-09-29:
> @italiangrandma: I just tested the games pictured. PAL games are already mentioned by agg23 to not be working properly due to waiting on updates from Analogye, so it shouldn't be logged as an issue here.

Ah yeah, gotcha. For some reason I thought that had been fixed in this release. I wasn't intending to "log it as an issue", but just adding context to this issue logged by Urbalcloud.

agg23, 2022-09-29:
> Edit: Just tested a handful of PAL games and can confirm the red line at the bottom appears in all of them.
I must have messed something up. This shouldn't be happening. If you open the core statistics, what number appears after "Sync OK"?

Urbalcloud, 2022-09-29:
Looks like it was a PAL rom. Converted to NTSC and it worked fine.

italiangrandma, 2022-09-29:
> > Edit: Just tested a handful of PAL games and can confirm the red line at the bottom appears in all of them.
> 
> I must have messed something up. This shouldn't be happening. If you open the core statistics, what number appears after "Sync OK"?

Hmmm, I'm not sure how to look at that...is it a button combo or something? I don't see any option for Statistics in the menus or anything to enable in the json files.

agg23, 2022-09-29:
It's a bug that anything shows down there, it's just black sometimes if the cores previously loaded black there.

italiangrandma, 2022-09-29:
> It's a bug that anything shows down there, it's just black sometimes if the cores previously loaded black there.

Ah, gotcha. I wonder if it happens reliably when I load an NTSC rom that has black in that area and maybe when I quit the core while that black part is still on the display...

agg23, 2022-09-29:
It looks like it's complaining because I didn't update the aspect ratio to account for the changed values. Can you try this:

```json
{
  "video": {
    "magic": "APF_VER_1",
    "scaler_modes": [
      {
        "width": 512,
        "height": 224,
        "aspect_w": 8,
        "aspect_h": 7,
        "rotation": 0,
        "mirror": 0
      },
      {
        "width": 512,
        "height": 224,
        "aspect_w": 4,
        "aspect_h": 3,
        "rotation": 0,
        "mirror": 0
      },
      {
        "width": 512,
        "height": 240,
        "aspect_w": 16,
        "aspect_h": 15,
        "rotation": 0,
        "mirror": 0
      },
      {
        "width": 512,
        "height": 240,
        "aspect_w": 4,
        "aspect_h": 3,
        "rotation": 0,
        "mirror": 0
      }
    ]
  }
}
```

These aren't the best possible aspect ratio values; I just want to see if it fixes it. I think 4:3 is probably still broken, but the 16:15 should be right...

agg23, 2022-09-29:
Ok, thanks to @italiangrandma, I figured it out. PAL games can either be 240 or 224px tall, and I only tested 240 tall games, hence the sudden issues.

This will be resolved in the next update