agg23/openfpga-SNES #41: Kyuuyaku Megami Tensei is Crashing possibly.

The-grey-ghost, 2022-09-30:
When attempting to use the core it shows the Atlus logo but after attempting both letting it play and pressing any button to skip, it leads to a black screen where no audio plays, not even a hanging note.  Attempted to use both an English patched version and then unpatched version to same result, no saves or anything attempted to be brought over. The analogue button still works and allows me to quit out.

agg23, 2022-09-30:
I cannot repro. I did not try the English patch, but the original Japanese ROM ran just fine

The-grey-ghost, 2022-09-30:
alright possibly the rip I'm using, guess I got to look around and I'm using the basic patch tools for roms to apply English patch

agg23, 2022-10-01:
Closing for now. Please reopen if you can't fix it

closed 2022-10-01 by agg23

The-grey-ghost, 2022-10-02:
Reripped and japanese works as well and whatever the IPS patch does I guess makes it bug out.