agg23/openfpga-SNES #43: Can't Get Far East of Eden To Load

DarkAcid112, 2022-10-02:
I've seen people getting this to load, but not having luck on multiple roms. What am I doing wrong?

SegaSnatcher, 2022-10-02:
The current build doesn't have SPC7110 support yet.   You'd have to use a custom build. 

You can use this as your Far East of Eden core for now, but don't get rid of the main build.  You can switch between the two cores.  

Just unzip this file and add it to your core folder. 

[agg23.SNES-SDD1.zip](https://github.com/agg23/openfpga-SNES/files/9693248/agg23.SNES-SDD1.zip)

DarkAcid112, 2022-10-02:
Looks like I can get it to run the SPC7110 check program, but getting the NG after every check using the original J rom. When I use the patched English version, I still get a black screen. I guess I’m doing something wrong.

SegaSnatcher, 2022-10-02:
Oh apologies, I forgot that game requires a save file to be placed in the save folder for it to work.   

Just place these in the SNES save folder.  But make sure they follow the same file path as where the rom is placed in your assets.  Also, they need to match the name of the rom as well. 

[Saves.zip](https://github.com/agg23/openfpga-SNES/files/9693309/Saves.zip)

DarkAcid112, 2022-10-02:
OK, so I have the save files placed in the SNES save folder, and the path is the same where the rom is placed in the assets path. It also has the same name, but I’m still getting the black screen. I chose the SDD1 core as the default as well. Currently on the latest Pocket firmware. Am I missing something?

SegaSnatcher, 2022-10-02:
Try this build.  

[agg23.SNES-SPC7110.zip](https://github.com/agg23/openfpga-SNES/files/9693445/agg23.SNES-SPC7110.zip)

DarkAcid112, 2022-10-02:
That seems to get me further. It fails on RTC Backup and says reset the system.  Thank you for helping me with this.

SegaSnatcher, 2022-10-02:
That is unfortunate as it works for me without issues.   Not sure what could be the issue.

DarkAcid112, 2022-10-02:
Thanks anyways. I’ll keep messing around with it.

SegaSnatcher, 2022-10-02:
Are you on discord?   If you care to, you can join my server and I can provide everything I have.   

https://discord.gg/jkBQnjTb

DarkAcid112, 2022-10-02:
Thanks for the help. Issue is now resolved.

closed 2022-10-02 by DarkAcid112

DarkAcid112, 2022-10-11:
Awesome! Thank you.