agg23/openfpga-SNES #44: Super Starwars fails to load after end of first level

Orbitoid, 2022-10-03:
Game loads OK but when finishing the first level, the game locks up after the points/time bonus have finished counting down (Stage Clear screen). Cannot get past this screen. 
This happens on release 0.3.1 with and without headers on the ROM.

Orbitoid, 2022-10-03:
It seems to get stuck just before it tries to do the "circle fade out" (Black circle starts at the center of the screen and expands to fill the screen to progress to the cut scene)

agg23, 2022-10-27:
I'm not good enough at the game to verify this. Have you tried a different ROM?

GreatHierophant, 2022-11-11:
Works fine for me.

Orbitoid, 2022-11-13:
Just updated and also changed rom at the same time. Looks to be working for me now. Not sure if it was the ROM or something in the latest update but it is working fine now.

closed 2022-11-13 by Orbitoid