agg23/openfpga-SNES #56: Delete save when reset core and CPU turbo are enabled together.

multi-vitamin, 2022-11-17:
If you use cpu turbo and choose reset core instead of load cartridge,
The save will be deleted.

i checked this happen with Ys3, Final Fantasy 6.

However, Ys4 was not deleted.

Perhaps if the glitch is severe, it will be deleted.

agg23, 2022-11-17:
Do either of those titles actually experience slowdown?

[From the MiSTer turbo docs](https://github.com/MiSTer-devel/SNES_MiSTer/blob/master/SNES_Turbo.md):
> Use of the CPU Turbo function is outside the normal operation of SNES hardware. Undesirable issues will occur if you use this feature in many games.

The save was not deleted, it was corrupted. Did the game actually function besides the save corruption?

multi-vitamin, 2022-11-17:
Game didn't slow down, it was just a test.
It doesn't support it, so only a glitched screen comes out.

Any ideas why save file gets corrupted when I select core reset option?

agg23, 2022-11-17:
If you enabled the turbo option after the game started the first time, then resetting the core would be the first time the game started from its initial state with the turbo enabled. If that's not the case, it's possible that whatever corruption occurs to the game executing state clobbers the reset vector, so specifically using the reset option causes the CPU to jump to that vector, which is where things get messed up in the save.

Since turbo messes with the timing of the CPU, anything could be happening, so somehow the save RAM is getting clobbered.

In any case, there's nothing I (or anyone really) can do here. When you mess with CPU timing, anything can happen.

multi-vitamin, 2022-11-17:
Thank you for reply.

closed 2022-11-17 by multi-vitamin