agg23/openfpga-SNES #58: [Function request] Is Save States possible?

MelonGx, 2022-11-25:
I plan to play SMW hacks on AP but it seems that AP’s Save State doesn’t work on OpenFPGA cores.

agg23, 2022-11-25:
No, save states are not going to happen. It takes a LOT of work for FPGAs, almost akin to making a separate core. If the original core doesn't support it, it's highly unlikely to come to the Pocket.

closed 2022-11-25 by agg23

DunhamGitHub, 2023-03-10:
I'd love to see this feature.
Tbh, the APocket is useless to me without it.

_...I want to turn it off and have it resume when I turn it back on, I just cannot be bothered to "search a save point" (it's 2023 ffs🤭)_

* I'd contribute an easy 100$ to see this done, probably many would too. Maybe a kickstarter-campaign for this exact feature might be in order