agg23/openfpga-SNES #59: Sleep Mode support?

zkat, 2022-12-09:
When I try to do sleep mode on my Analogue Pocket, it says it's not supported by this cartridge, and offers to power off instead. Is it possible to add support for sleep?

agg23, 2022-12-09:
Sleep requires save states, which won't be coming to the SNES core. The only third party core currently available with save states is my NES core.

zkat, 2022-12-09:
Thanks for the response! I was wondering whether this was a save state thing. I've linked this issue (see above) in upstream in hopes that it'll get resolved someday. :)

DunhamGitHub, 2023-02-03:
For me, save states / sleep mode is essential for SNES. Which is why my analogue pocket is collecting dust...

Hoping it will happen due to competition in the market (I see it on the horizon)

ToniWonKanobi, 2023-03-27:
Just wondering why save states won't be coming to this SNES core? Is that the owner doesn't care, or that it's incredibly difficult (more difficult than NES)?

agg23, 2023-03-27:
Unlike software emulators, hardware emulators (what we're running on FPGAs) have a significantly higher complexity and must keep track of much more information. Consequently, retroactively adding save states to a core that was not planned around it is a very difficult task (and it's still not easy if you have planned for it).

The person most knowledgeable about the core, srg320_, has previously stated that he has no interest in adding save states. Unless someone very skilled and dedicated decides to do it, it isn't going to happen. It's honestly probably more likely that someone else starts a SNES core from scratch with the idea of including save states.

ToniWonKanobi, 2023-03-27:
> Unlike software emulators, hardware emulators (what we're running on FPGAs) have a significantly higher complexity and must keep track of much more information. Consequently, retroactively adding [...]

Thanks for the quick reply. I hear what you're saying, but why does the NES core have save states then?

ToniWonKanobi, 2023-03-27:
Fair enough. Really appreciate your responses. Hopefully, my questions didn't come off as frustration pointed at your direction. Thanks, and have a good day sir!

zarkone, 2024-11-20:
An update in MISTer land about save states: https://misterfpga.org/viewtopic.php?t=8459

agg23, 2024-11-20:
> An update in MISTer land about save states: https://misterfpga.org/viewtopic.php?t=8459

It's a very impressive set of changes. I don't believe the source is available at the moment, and even if it was, I don't think it's in a stable enough state to warrant the significant amount of work it would take to get working on the Pocket. I will look into it as it becomes more stable and I have time available though.

ToniWonKanobi, 2024-11-20:
> It's a very impressive set of changes. I don't believe the source is available at the moment, and even if it was, I don't think it's in a stable enough state to warrant the significant amount of [...]

Does this mean save states could be coming to the SNES core!? 🙃

TheJayMann, 2024-11-24:
I've not done anything to verify this outside of compare names and dates, but I believe the source code for the latest release in the forum thread (including the update for SA1 support) is available at https://github.com/paulb-nl/SNES_MiSTer/tree/ss_wip.

DragonPokeBall, 2025-03-10:
Just wait till someone will make another core that will have savestates/sleep/memories. This person was too lazy to fix it.

agg23, 2025-03-10:
> This person was too lazy to fix it.

That is a very unhelpful comment. I have literally stated that once it's fairly stable I will look into it. As of the moment it breaks in a bunch of scenarios.

> I don't think it's in a stable enough state to warrant the significant amount of work it would take to get working on the Pocket. I will look into it as it becomes more stable and I have time available though.

I do this in my free time and am not beholden to your whims on when I should finish this. This takes a very large amount of work and skills that few possess. I would happily accept your PR adding the feature.

zarkone, 2025-03-10:
@agg23 I hope it is rather an exception to get this type of comments.. I also want to assure you that majority of people in the scene understand the complexity and the value of your labor here 💟 

tbh, it feels like it might be even impossible to add this thing without complete rewrite of the core. Even with my very limited knowledge, I understand that mister != analog fpga and adding saves to a core which was designed without saves in might might be impossible, because 

> a very large amount of work and skills that few possess

very sad to see such feedback :( your core is awesome and I'm using it with pleasure 🙏 

Thanks for your work!

ToniWonKanobi, 2025-03-10:
> Just wait till someone will make another core that will have savestates/sleep/memories. This person was too lazy to fix it.

Haha. This is pretty harsh, but I do hope the agg23 dude can add save states at some point 🤞

Skyyblaze, 2025-03-11:
I also want to extend my sincere thanks to you agg23, I'm no developer but I understand what a daunting task such a FPGA core must be. Playing SNES on my Pocket became my favorite way to play SNES games thanks to you!

fox1t, 2025-04-08:
@agg23 thanks for your work. I am an OSS maintainer too and I know how hard it can be sometimes. <3

zkat, 2026-03-30:
@ToniWonKanobi Analogue probably won't. I hear agg23 isn't super available lately and porting things to the Pocket isn't exactly easy, so we'll need to wait until someone available and capable comes along to port the suspend-capable thing over.