agg23/openfpga-SNES #62: Star Fox EX 1.04.3 blank screen

DJBabyBuster, 2022-12-15:
Just updated the core, and the latest Star Fox EX patch, and I'm getting a blank screen for 1.04.3 with sound.  Version 1.01 works fine.  Just looking to see if you can confirm if 1.04.3 is working, thanks

agg23, 2022-12-15:
Are you using the patch without FastROM? The FastROM version doesn't work. See https://github.com/MiSTer-devel/SNES_MiSTer/issues/320

DJBabyBuster, 2022-12-15:
Hi yes using without FastROM, with Star Fox (USA) (Rev 2). I was stumped because as I said patch 1.01 works fine, but the latest 1.04.3 will not load video on my pocket with the the latest snes core 0.4.3

DJBabyBuster, 2022-12-15:
Okay tried the "SFEX (HARDWARE) (FOR MISTER) 1.04.3.bps" and it's working! Apparently just "SFEX 1.04.3.bps" doesn't work for pocket

agg23, 2022-12-15:
Glad you figured it out. For some reason some of the patches don't work on MiSTer (and by extension on Pocket).

closed 2022-12-15 by agg23