agg23/openfpga-SNES #63: Error in framework Can't find Chip32 blob

delpech, 2022-12-16:
Hello,

more likely a problem coming from me but I can't find what I'm doing wrong.

I have the error message "Error in framework Can't find Chip32 blob""

I use .sfc roms.
And copy the files as follows:
- snes in /Assets
- agg23.SNES in /Cores
- snes.bin in /Platform/_images
- snes.json in /Platforms

Any idea what I am missing ?

Kind regards,
Sylvain

agg23, 2022-12-16:
You didn't copy the core files over correctly. Download the `agg23.SNES_0.4.3_2022-12-14.zip` file in Releases, and make sure you have a `snes_main.rev` and `loader.bin` in the `/Cores/agg23.SNES` folder

delpech, 2022-12-16:
It works, thanks! I had downloaded the master repository (openfpga-SNES-master.zip) instead of taking a release.

closed 2022-12-16 by delpech