agg23/openfpga-SNES #65: Romhack issues

DrAvedi, 2022-12-24:
thanks for continuing to maintain this port! It's been my most-used core on the Pocket, and I bet Analogue won't release a SNES core, as it would compete with their Super NT.

anyhow, I was just trying out some hacks and they won't run properly. The hacks are Star Fox EX [https://www.romhacking.net/hacks/7285/](url) and Ultra ClayFighter [https://www.romhacking.net/hacks/6807/](url). Initially, I thought there was an issue with the patcher I was using, but the roms run in an emulator just fine. I haven't tested on real hardware.

ClayFighter freezes before the fighters appear in the Game Start mode.

Star Fox EX has significant graphical issues.

agg23, 2022-12-24:
See #62 for Star Fox EX. You need a SlowROM version; I believe there's a build specifically for MiSTer that should work.

Do you know if the Ultra ClayFighter hack runs on MiSTer? Most likely it has the same problems there.

DrAvedi, 2022-12-24:
oh, got it. I'll check out the bps for MiSTer, as I did see it in the zip for the hack.

I don't own a MiSTer to test ClayFighter. Might just be a foregone thing for now.

agg23, 2023-04-19:
For Ultra ClayFighter, MiSTer does not freeze at the same location, but it does indeed freeze at some point. Does Ultra ClayFighter work as expected on original hardware?

Unless we establish there is something unique to this port, I'm going to close the issue for now.

closed 2023-04-19 by agg23