agg23/openfpga-SNES #66: Roms load to black

joekwon, 2023-01-07:
**Environment:**
openfpga-SNES  0.4.3
Analogue OS v1.1 b6

**Problem:**
All roms fully load (white progress bar at bottom goes all the way across), hear a pop of sound, then remain black. 

**Attempted Troubleshooting:**
I've tried both headered and headerless roms and all have the same behavior. Also tried multiple SD cards, formatted both FAT32 and exFAT, both on Windows & Mac. All yield same results. 

Happy to provide any additional information if needed.

agg23, 2023-01-07:
That's really weird. Do you have any of the other cores working? To verify, your `/Cores/agg23.SNES/` folder contains `snes_main.rev` and other `rev` files, correct?

Just to check for some regression that may be specific to something with your hardware, try some of the older core versions

agg23, 2023-01-07:
GBA sounds like you're missing the correct BIOS.

There shouldn't be anything with a given device's hardware that presents problems with cores, but we have noticed some strange behavior on certain builds of certain cores, which is why I had you try multiple versions.

Did you copy the files directly over to the SD card, rather than using the USB option? Can you post a picture of your file structure just so I can verify it? Have you tried using one of the updater tools described in the README or release notes?

It shouldn't really make a difference, but delete your `/Saves/snes` folder to make sure games aren't loading some weird corrupted save.

agg23, 2023-01-07:
Everything looks fine. I'm not sure what else to suggest. If you haven't already, try setting up a new SD card with _only_ the SNES core.

It's also worth checking that _my_ NES, PCE, Arduboy, etc cores work properly.

agg23, 2023-01-08:
> When only the SNES core is on the SD card, the openFPGA option shows the openFPGA Guide QR Code, as if the core is not loaded. However if I load another core onto the SD card, then that core plus the SNES core will show up. For some reason it's not being recognized when it is by itself.

That suggests something is indeed wrong in either the Pocket's OS, or the SNES files. The files of interest for showing in the UI are `/Cores/core.json` and `/Platforms/snes.json`. If either of those files are messed up, the Pocket may do strange things.

agg23, 2023-01-08:
I'm sorry, yes it's `/Cores/agg23.SNES/core.json`

ricjd, 2023-03-05:
@joekwon did you hear back from pocket, I'm getting the exact same issue, i.e. only SNES roms won't work and just load to black.  I don't have another Pocket to confirm sd card though

ricjd, 2023-03-05:
Yeah GBA games work.  Just sees games.  Been trying everything, like other SD cards etc but no luck.

agg23, 2023-03-05:
I would really like to know what causes these failures on the occasional device. There should not be differences in how a particular FPGA chip handles these cores, but many reports I've seen suggest that may be the case. The other common problem, is the SD card, which has already been mentioned.

@joekwon if you're willing, can you share more information about what support said and what they determined needed to be repaired? You can privately email me if you prefer. I'm wondering if they actually identified a problem, or if they were unable/unwilling to view an openFPGA issue as an actual support issue, and therefore wanted to charge you some fee to discourage it.

ricjd, 2023-03-05:
@agg23 I've ordered another sd card, just to try as wondering if that's the issue here. If you need anything from me, more than happy to help debug.  I tried turning on logging to see if I could see anything, but nothing jumped out