agg23/openfpga-SNES #67: Aspect ratio incorrect / missing

xeromid, 2023-01-09:
First of all, thanks for your ongoing work on this core! 

I have, however, discovered an issue with the video output. 
The aspect ratios available in the core (with adjustment via the _Square Pixels_ option) are not correct. 

The SNES has two aspect ratios: 8:7 and 4:3. This is because the development hard- and software provided by Nintendo worked with (and showed) an 8:7 ratio (the Super Nintendo's internal resolution). When the games were displayed on a TV set (on the consumer side), the image was usually stretched to fit the TV's 4:3 screens. So, while many SNES games are displayed correctly with an 8:7 ratio (e.g.: Super Mario World & Super Mario Allstars), some developers incorporated the subsequent stretching on the consumer side in their graphics, thus producing their games to be displayed correctly in 4:3. These ratios, 8:7 and 4:3, are actually quite different. They don't "roughly correspond", they need options to be displayed separately by all means.  

When the _Square Pixels_ option is unchecked, the output (on the Pocket's display, I don't own a dock) is definitely not the Super Nintendo's "native" 8:7 ratio: The image is clearly stretched horizontally. You can notice this very easily when you start Super Mario Allstars. The Mario coin shown in the beginning would be perfectly round on a correct 8:7 ratio. Just compare to the same game running on a Super Nt in 8:7 aspect ratio. 

As far as I can see, with _Square Pixels_ unchecked, the games are displayed with, or quite close to, a 4:3 ratio. 

However, with _Square Pixels_ checked, it doesn't work as 8:7 as well. Here the image is stretched vertically, almost filling the Pocket's 10:9 display. An aspect ratio not used in any games on the SNES, as far as I know. 

So what needs to be incorporated are correct output options for both 8:7 and 4:3, so any game can be displayed as it is intended to.

agg23, 2023-01-09:
Please see #36. We've already discussed this at length.

My understanding:
The native SNES resolution is 8:7 (PAR). Displayed on a 4:3 CRT, this comes very, very close to a 64:49 DAR (4:3 is 64:48, where the extra 1 is accounted for in overscan). 8:7 DAR is not something that you would ever see (same with 1:1, or square pixels).

xeromid, 2023-01-09:
Luckily in 2023 we're a step or two further from just replicating what was displayed on a CRT, as that wasn't how many games were actually made for. Just refer to the pictures on top of that thread and you'll see it makes quite a difference: 

[https://videogameperfection.com/forums/topic/43-87-aspect-ratio-correction-for-snes/](url) 

As you see, I think being able to incorporate a choice between 4:3 and 8:7 is very important - this is not about splitting hairs over technical details. Forcing to 4:3  only (I don't get what the square pixel option would be good for) means a slew of games are going to run clearly with incorrect aspect ratio. 

Of course, for some people (see #36) it only matters what the physical consoles were, or are (I too still have my SNES from the 90s) putting out. Fine with me - stay with the one option you like. But in my view it's a must to be able to play Super Mario World in the correct aspect ratio! Just look at the tubes - it's all wrong in 4:3!

RaspberryAlpine, 2023-01-09:
Xeromid, the core already does what you want it to do. The default aspect ratio setting is the aspect ratio a real snes would output on a 4:3 CRT. The square pixel setting is the 8:7 aspect ratio (for 256 by 224 resolution games).

SegaSnatcher, 2023-01-09:
I concur, the core already has both options available.

xeromid, 2023-01-10:
> Xeromid, the core already does what you want it to do. The default aspect ratio setting is the aspect ratio a real snes would output on a 4:3 CRT. The square pixel setting is the 8:7 aspect ratio (for 256 by 224 resolution games).

As I have already pointed out, it doesn't. While the default setting supplies a correct 4:3 picture, the Square Pixels option does _not_ give a correct 8:7 aspect ratio, the image is stretched vertically, slightly, but undeniably. Just check any round or square objects in 8:7 games (like, as I mentioned, the Mario coin displayed in Super Mario Allstars on bootup, ), it is clearly visible. Here a correction is needed, don't you agree?

RaspberryAlpine, 2023-01-10:
This is from the video.json file:

      {
        "width": 512,
        "height": 224,
        "aspect_w": 8,
        "aspect_h": 7,
        "rotation": 0,
        "mirror": 0
      },

As you can see the aspect ratio is indeed 8:7. If isn’t 8:7 on the Pocket’s then the problem isn’t with the core but with Analogue’s scaler.

RaspberryAlpine, 2023-01-10:
This is the Super Mario All Stars boot up coin on my Pocket:

https://media.discordapp.net/attachments/335174299306688514/1062415967520559104/IMG_1699.jpg

The coin measures the same vertically and horizontally on my pocket.

It is possible you don’t have the latest video.json file. It is located in /Cores/agg23.SNES/video.json

Riptides, 2023-01-10:
I can confirm I observe the same behavior as @RaspberryAlpine

RaspberryAlpine, 2023-01-11:
@xeromid can you confirm the issue is still happening with all the latest core files, especially the video.json file?

xeromid, 2023-01-16:
I have had the wrong video.json. You have my apologies! 😔

closed 2023-01-16 by xeromid