agg23/openfpga-SNES #68: 2.4g wireless controller bad connection

robertvb83, 2023-01-21:
Don't know if this is a core issue or an analogue one.

With cartridge games i can connect my 8BitDo 2.4g wireless gamepads (2 of them) without any issues.

However in openFPGA e.g. with this SNES core, i have a very bad and unstable connection. The controllers are disconnected all the time during gameplay. I read about that on reddit, so others also have this issue.

With bluetooth controllers everything is fine for both, Cartridge play and openFPGA SNES core.

agg23, 2023-01-21:
This is an Analogue issue. I am interested in whether or not you notice the connection being unstable in other cores (either openFPGA or built in).

robertvb83, 2023-01-21:
With built-in cores (gba, gbc, GG cartridges) the connection is very stableWith openFPGA it is very unstable (i mostly play SNES, so i have to check other cores, and report back)

  (Issue #68)

robertvb83, 2023-01-24:
A workaround would be to use Bluetooth controllers. You can use 4 in total by connecting two directly to the dock and 2 via 8BitDo wireless usb receiver 2 dongle. However the controllers are not always recognized correctly. Something is fishy.

The downside with bluetooth however is a lot more noticeable lag

It is like someone on Reddit already said: NES never disconnects. GBA sometimes, and SNES and genesis all the time.

Analogue tells us to turn off sources of interferrence like 2.4g WiFi and Zigbee lighting, which is not really realistic

For reference
https://www.reddit.com/r/AnaloguePocket/comments/y9gbrx/anyone_elses_dock_8bitdo_24g_controllers_still/

https://www.reddit.com/r/AnaloguePocket/comments/103ftim/8bitdo_sn30_24g_controller_randomly_disconnecting/

https://www.reddit.com/r/AnaloguePocket/comments/rqlgrc/24g_controller_disconnecting_repeatedly_from_dock/

robertvb83, 2023-01-25:
@agg23 i hope its ok to "misuse" your github issue as a tracking for this 8bitdo/analogue related issue.

I try to concentrate the info here and link 8bitdo / analogue support to this entry too.

agg23, 2023-01-25:
Yea it's fine. I've reached out to Analogue to see if they have any additional info, but I really doubt that it actually differs by core (unless they did something very wrong).