agg23/openfpga-SNES #73: Retro-Bit SNES Data East Classic Collection doesn’t work

fmdoppel, 2023-03-01:
Game Boot, but there are some graphical glitches in the Menu. At least you can guess which Game from the collection you select. 

If you do so, the Game simply freeze.

agg23, 2023-03-01:
Have you checked against MiSTer? Most likely it doesn't work there either, and you're more likely to get traction if you [report it to their repo](https://github.com/MiSTer-devel/SNES_MiSTer).

fmdoppel, 2023-03-01:
I haven’t checked it, die I don’t own one. 

Today I just tried "Retro-Bit SNESJoe & Mac: Ultimate Caveman Collection" and "Retro-Bit  SNES Jaleco Brawlers Pack".
Both of them have exactly the same Problem. I don’t own the R-Type Collection from them so I can just guess that the same Problem occurs here. 

Thanks for the Link @agg23, I’ll report it there also.