agg23/openfpga-SNES #74: A-Button doesn't work

codeandcreate, 2023-03-04:
The right button "A" doesn't work in all games.
It works in other cores or in the core menu. but not in games.
Is this general problem?!

agg23, 2023-03-04:
That is really weird. It works correctly for you in other openFPGA cores?

I would mess with the various required files. Delete the core and copy it again, try another SD card, etc. We've found the Pocket is very sensitive to SD and filesystem issues, though I haven't seen cores partially work as of yet.

codeandcreate, 2023-03-04:
Yeah it works on all other cores, cardrige games (...) on everything else ... i will try to reinstall.

agg23, 2023-03-05:
> Maybe something has changed in the settings config from 0.4.2 to 0.4.3 ? After i killed the settings json it works again ô.o
Interesting. No, nothing should have changed settings-wise between any of the 0.4 versions.

Did you remap your controls at any point? This sounds like it could be related to that.

closed 2023-03-15 by agg23