agg23/openfpga-SNES #75: Project Nested (1.7) ROMs have no sound

SpiralBrad, 2023-03-15:
This differs from the current MiSTer core, as of the date of this post.

[Super Mario Bros. (World).nes.smc.zip](https://github.com/agg23/openfpga-SNES/files/10984903/Super.Mario.Bros.World.nes.smc.zip)

agg23, 2023-03-15:
Interesting. I think this is the first confirmed discrepancy against MiSTer.

Since it's audio, I wonder if the internal SNES core is producing audio as we would expect, but it's not making it out over the Pocket's i2s bus for some reason

SpiralBrad, 2023-03-15:
One further thing of note, on ONE short playtest of the posted ROM, the Pocket did spit out some gnarly static during gameplay / until switching ROMs, but I can't seem to replicate it.  Definitely incorrect either way ;)

agg23, 2023-03-15:
By static, do you mean there was some garbage static audio? Not the screen was full of static?

SpiralBrad, 2023-03-15:
> By static, do you mean there was some garbage static audio? Not the screen was full of static?

Correct; audio static. Sorry, should have been more clear.

agg23, 2023-04-19:
I need to look into this more, but I wanted to note that I have also confirmed the audio issue. Working on MiSTer, nothing on Pocket.