agg23/openfpga-SNES #80: Issues with SPC7110 Support - Far East of Eden

cwake30, 2023-05-11:
So I just got my pocket and updated to the newest firmware and then downloaded the auto updater that downloads all of the cores and such for my device, loaded games into the appropriate folders on my microSD card and I have seen on videos and read (on here also) that the SNES core has support for RTC/real-time clock as well as all of the special enhancement chip games (SuperFX, CX4, DSP, SA-1, etc, etc) including the SPC7110 chip which allows Far East of Eden ZERO (Tengai Makyou) to be played. So I put the game on my SD card and tried to run it on the Pocket and there are two SPC7110 check screens, the first said ALL OK, Please reset, so I did, then it runs a second check (on the S-RAM and on the RTC backup) and it says:

RTC Backup - NG

S-RAM Backup - OK

Please reset, but doesn't give the ALL OK message. At that point it just loops and fails that RTC part and tells me to reset it over and over and over. I REALLY want to enjoy the support for this game and play it, thank you for any help in advance!

agg23, 2023-05-11:
The initialization of this game is super weird, and we wondered about this when initially testing, but we concluded that everything is probably correct.

Easy solution, just replace your Far East of Eden save with a known working one. Otherwise:

1. Delete your save file from your SD card
2. Open the game. It will prompt you to press A
3. Press A, and when it says "All OK", open the core settings, and press "Reset" then close the menu
4. The game should now prompt you to press B
5. Press B, and when it says "All OK", open the core settings, and press "Reset" again
6. Your game should now be functioning

cwake30, 2023-05-12:
Thanks so much! I got It to restart the checks from the beginning and finally It gave an ALL OK on both and I reset and the game started up. Very happy!

closed 2023-05-12 by cwake30