agg23/openfpga-SNES #81: .FIG and .SWC files not showing in list

Slashspade, 2023-08-28:
As the title suggests, my snes/super famicom roms with .fig and .swc extensions aren't being detected by the core/device. Any idea what's going on?

agg23, 2023-08-28:
I have never heard of either of those formats. It looks like they are both from proprietary dumpers that are quite old now. I was unable to determine whether or not the format is actually the same, or if these ROMs are somehow different than `smc` or `sfc`.

First, you probably don't want to use those ROMs. They're very old and may not be accurate dumps. You should use verified accurate ROMs instead.

Now, assuming you're not going to replace your ROMs, you can open the `/Cores/agg23.SNES/data.json` file and add the `fig` and `swc` extensions to the list for the `Cartridge` slot. You could also rename your ROM files to be `*.smc`.

Please report back on whether or not the ROMs work in the core. If so, I can push an update with that `data.json` change.

Slashspade, 2023-08-29:
So I did what you said. The .fig files were found, but the .swc weren't. I changed the extension on that one like you said and then it worked. I'm no expert to any of this - barely a novice really- so I wasnt about to go and change file extensions without someone else suggesting it first.

agg23, 2023-08-29:
Ah sorry. I just checked, and there's a maximum of 4 supported extensions in that list. So the last one you added didn't count. I can't do anything about that unfortunately.

Slashspade, 2023-08-29:
I gotcha. That makes sense, that was the last one in my list. But it's not a huge deal - it's working now.