agg23/openfpga-SNES #82: MSU-1 Support

briaguya-ai, 2023-08-29:
This was requested in a [reddit thread](https://www.reddit.com/r/AnaloguePocket/comments/y7qppe/msu1_support_for_snes_core_for_analogue_pocket/), where @agg23 [responded](https://www.reddit.com/r/AnaloguePocket/comments/y7qppe/comment/iswck6j/?utm_source=share&utm_medium=web2x&context=3)
> MSU requires streaming, which the firmware does not currently support. I’ll work on it once that update happens, though I expect some difficulty.

I figured it was worth making an issue on the repo to have this documented somewhere other than reddit.

briaguya-ai, 2023-08-29:
> I personally have 0 interest in the feature, and given the large amount of work it represents, most likely will never take it on.

That is completely understandable.

> if someone else wishes to undergo the process

For any potential "someone else": here's a link to the PR adding MSU-1 support on the MiSTer https://github.com/MiSTer-devel/SNES_MiSTer/pull/328

NuVanDibe, 2023-12-14:
I'd like to throw my wish into the ring for this as well. MSU-1 is an incredible addition to the romhacking toolkit, and it's given us some incredible content.

Sadly I know nothing about FPGA development.

thehughhefner, 2024-06-24:
Hi,
Any updates on this.
This is one of the features that is missing from MiSTer

agg23, 2024-06-24:
As stated above

> this is solidly in the won't happen category

unless someone else picks it up.