agg23/openfpga-SNES #83: Kaizo Hack Passcode Mechanism not working

gizmo-321, 2023-08-30:
Hi there,

i tried to play the SMW-Hack Chain Reaction (https://smwc.me/s/18669) and the Creator uses a Passcode-System to go to the Level Selection. You have to jump on each box to count them up. But this doesn't work. The Box stays at zero. I jumped also on them and count by myself, but this also doesn't work. Maybe it's possible that your core is missing a feature to be able to work with this passcode system, but i don't know technical details how this PAsscode System works.

Same Rom work fine on the SuperNT and FXPakPro...

[image]

agg23, 2023-08-30:
I'm guessing the hack is emulator tested and not necessarily accurate for hardware. There's a [SMWCentral comment](https://www.smwcentral.net/?p=section&a=details&id=18669&r=0#comment-32370) that indicates it doesn't work on BSNES or Higen, which are the two highest accuracy emulators.

Ultimately this is an issue for upstream at https://github.com/MiSTer-devel/SNES_MiSTer, but without evidence of the hack not working properly on MiSTer, they most likely will not take kindly to you opening an issue there.

gizmo-321, 2023-08-30:
Oh i totally overseen this comment. I tested and i had to jump 105 times till the first box started counting. I know a guy that owns a MiSTer. I will ask him if this hacks works correctly :)

gizmo-321, 2024-01-19:
In the meantime I bought myself a MISTer and tested this issue, and it is also present there. I will open an issue as suggested above ;-)

gizmo-321, 2024-01-20:
Take look at https://github.com/MiSTer-devel/SNES_MiSTer/issues/373#issuecomment-1901981786 there is the solution for the MISTer, but this setting is not available on your core ;-)

closed 2024-01-20 by gizmo-321