agg23/openfpga-SNES #84: Nyquist filter request.

Whosondephone, 2023-09-06:
Firstly thank you for this core.  It's truly magnificent!  My only problem with it is the Nyquist filter.  It sounds like the filter is set to compensate for a sampling rate of 44.1kHz as apposed to the SNES's 32kHz.  This produces a distortion known as aliasing.  It produces a sound similar, but not as bad, as the GBA.  Because I assume that the GBA doesn't have a filter and runs at 22kHz if memory serves.  

I'm assuming that there is no filter built into the core and we are relying on the Pocket's dac to supply the filter via analog circuitry.  No pun intended.  

I can see how people may prefer the current sound.  Heck, I kind of like it.  It sounds brighter.  But can be distracting to me at times.  And is not accurate to how the SNES sounds.  This issue may come from the original MiSTer core as well.  I wouldn't know as I don't have access to one.

Whosondephone, 2023-09-09:
I've been scratching my head on this one too.  It's such a simple solution in an analog circuit.  Just throw a capacitor in parallel or a coil in series.  But we don't have that option in FPGA.  The other thought I had was to change the output of the sound system to 48kHz.  I can imagine reworking the logic to make the piches correct.  I actually may take a stab at that if no one gets around to it in the next 20 years.  (No sarcasm).  I'm surprised to realize how I've been taking Digital Signal Processing for sound for granted.  I have no idea how how a basic EQ DSP works.  Time to educate myself.

Can you elaborate on this default set of filters?  Are they in the FPGA or in the CPU in mister.  What do you mean by EE?

agg23, 2023-09-09:
Sorry, EE is electrical engineering. I'm all self taught, and while I have a math background, I don't understand most things about electronics.

I thought there were more built in filters than this, but this seems to be it: https://github.com/MiSTer-devel/Template_MiSTer/blob/2b795444020ed774a4e8b4560471c7ba50bbe0ca/sys/audio_out.v#L180-L222. The FPGA directly controls the audio on MiSTer; there's no control given to the CPU.