agg23/openfpga-SNES #85: A Link to the Past Practice hack

hungLink, 2023-09-12:
I know it's not an official rom, but it would be nice to know why this hack doesn't work, and if there is hope that it may work someday:
https://spannerisms.github.io/lttphack/
The game boots, and then after an initial sound (that is intended and good, even though it sounds alarming) there is a black screen, and the game doesn't seem to loag.

I asked in the ALttP practice discord and someone said "maybe they(agg23) limited SRAM to help it fit on the smaller FPGA?"
Is this the case?

In any case, the legacy version works just fine, and that's all I really need. But I'm still curious if anyone has any info.

agg23, 2023-09-12:
I can confirm the issue.

Maximum save size in the core is indeed set to 128KB, so that could be the problem. Do no other games use the largest possible save size? This is technically an upstream issue for the MiSTer core.

hungLink, 2023-09-12:
I shared this thread with the practice hack modding community and they said it should still technically work, but I wouldn't have savestates, because those two RAM banks would be mirrors of the other save data.

How complete is the SA-1 chip support on this core? That seemed to be a potential reason in their mind (given the old version works, and the new version utilizes this chip)

agg23, 2023-09-12:
I didn't write the core, it is a port of the upstream MiSTer core. You would have to ask them, as I don't know specifics of the development. There are very few, or no, retail titles that have problems with the core.

hungLink, 2023-09-12:
Thank you for taking the time to get back to me! You are a gentleman 😄

closed 2023-09-12 by hungLink