agg23/openfpga-SNES #91: When running a Chinese game, the text will be garbled

yujiegongzi, 2023-12-15:
Dear Sir/Madam,
I am a player from China. I can run Chinese games successfully and most Chinese characters display normally.  However, the text on the operation interface, items, and skill interfaces are confusing. 
And I have no problem playing Chinese games on Genesis and NES.
I don't know what is the reason for this.  Hope you can help me resolve my doubts.  Thanks.

agg23, 2023-12-15:
Can you post a picture of what is happening? Are these official games in Mandarin, or are they fan translations? Can you give me an example of a game?

yujiegongzi, 2023-12-15:
Thank you for getting back to me so quickly. Due to historical reasons,
Chinese SNES and SFC games are all fan-made Chinese versions, and there is
no official Chinese Snes game. I'll attach pictures below（Dragon warrier3,
Final fantasy5).

-- 

*PENAFIAN*: E-mel ini dan apa-apa fail yang dikepilkan bersamanya 
("Mesej") adalah ditujukan hanya untuk kegunaan penerima(-penerima) yang 
termaklum di atas dan mungkin mengandungi maklumat sulit. Anda dengan ini 
dimaklumkan bahawa mengambil apa jua tindakan bersandarkan kepada, membuat 
penilaian, mengulang hantar, menghebah, mengedar, mencetak, atau menyalin 
Mesej ini atau sebahagian daripadanya oleh sesiapa selain daripada 
penerima(-penerima) yang termaklum di atas adalah dilarang. Jika anda telah 
menerima Mesej ini kerana kesilapan, anda mesti menghapuskan Mesej ini 
dengan segera dan memaklumkan kepada penghantar Mesej ini menerusi balasan 
e-mel. Pendapat, rumusan, dan sebarang maklumat lain di dalam Mesej ini 
yang tidak berkait dengan urusan rasmi Universiti Kebangsaan Malaysia (UKM) 
adalah difahami sebagai bukan dikeluar atau diperakui oleh mana-mana pihak 
yang disebut.

_*DISCLAIMER *: This e-mail and any files transmitted with 
it ("Message") is intended only for the use of the recipient(s) named above 
and may contain confidential information. You are hereby notified that the 
taking of any action in reliance upon, or any review, retransmission, 
dissemination, distribution, printing or copying of this Message or any 
part thereof by anyone other than the intended recipient(s) is strictly 
prohibited. If you have received this Message in error, you should delete 
this Message immediately and advise the sender by return e-mail. Opinions, 
conclusions and other information in this Message that do not relate to the 
official business of The National University of Malaysia (UKM) shall be 
understood as neither given nor endorsed by any of the aforementioned._

agg23, 2023-12-15:
I think you have to upload the images in the Github UI, rather than replying over email.

yujiegongzi, 2023-12-15:
Ok, I will do that.

-- 

*PENAFIAN*: E-mel ini dan apa-apa fail yang dikepilkan bersamanya 
("Mesej") adalah ditujukan hanya untuk kegunaan penerima(-penerima) yang 
termaklum di atas dan mungkin mengandungi maklumat sulit. Anda dengan ini 
dimaklumkan bahawa mengambil apa jua tindakan bersandarkan kepada, membuat 
penilaian, mengulang hantar, menghebah, mengedar, mencetak, atau menyalin 
Mesej ini atau sebahagian daripadanya oleh sesiapa selain daripada 
penerima(-penerima) yang termaklum di atas adalah dilarang. Jika anda telah 
menerima Mesej ini kerana kesilapan, anda mesti menghapuskan Mesej ini 
dengan segera dan memaklumkan kepada penghantar Mesej ini menerusi balasan 
e-mel. Pendapat, rumusan, dan sebarang maklumat lain di dalam Mesej ini 
yang tidak berkait dengan urusan rasmi Universiti Kebangsaan Malaysia (UKM) 
adalah difahami sebagai bukan dikeluar atau diperakui oleh mana-mana pihak 
yang disebut.

_*DISCLAIMER *: This e-mail and any files transmitted with 
it ("Message") is intended only for the use of the recipient(s) named above 
and may contain confidential information. You are hereby notified that the 
taking of any action in reliance upon, or any review, retransmission, 
dissemination, distribution, printing or copying of this Message or any 
part thereof by anyone other than the intended recipient(s) is strictly 
prohibited. If you have received this Message in error, you should delete 
this Message immediately and advise the sender by return e-mail. Opinions, 
conclusions and other information in this Message that do not relate to the 
official business of The National University of Malaysia (UKM) shall be 
understood as neither given nor endorsed by any of the aforementioned._

yujiegongzi, 2023-12-15:
[image]
[image]
[image]
[image]
[image]

yujiegongzi, 2023-12-15:
Dragon Warrior3 and Final fantasy5.
I searched on Chinese websites and someone answered like this"In fact, it is not a problem with the simulator itself, but a problem with localization.  This Chinese version uses a bug in the early simulator, so it is also garbled on the actual machine.  Later, many simulators had options compatible with this bug, which needed to be turned on manually to display correctly."

yujiegongzi, 2023-12-15:
Due to historical reasons, Chinese SNES and SFC games are all fan-made Chinese versions, and there is no official Sfc Chinese game.

agg23, 2023-12-15:
> I searched on Chinese websites and someone answered like this"In fact, it is not a problem with the simulator itself, but a problem with localization. This Chinese version uses a bug in the early [...]

There's your problem then. FPGA implementations almost always are going to be as true to the real hardware as possible, and will not add hacks for incorrectly made games. If a game requires a certain emulator, it almost certainly will not work on FPGA.

As such, this seems to be a non-issue. Unfortunately you'll be unable to accurately play those hacks on Pocket.

yujiegongzi, 2023-12-15:
Thanks for your explanation. Unless it is re-Chinese, there should be no solution.