agg23/openfpga-SNES #92: Autotracking possible for playing randomizer

Quaschynock, 2023-12-16:
Hey, 
I love playing retro randomizer games on the Analogue Pocket. But is there a way to implement autotracking for items, locations, etc.? I usually play via the docking station and autotracking is the only thing I am missing. 

I do not even know, if you are able to do this or if it is a general feature in the Pocket or the docking station?? I heared about the new OS version and the openFPGA developer features. So I had hope again that this could be possible in the Future.

I really appreciate your project so thanks for all your work :)

agg23, 2023-12-16:
I actually discussed this with some people at Analogue. To do it, we would need a custom cartridge and a bunch of work to support the [USB2SNES protocol](https://usb2snes.com/). I'm not particularly motivated to do this, and it would be a lot of work, but it's certainly possible.

Quaschynock, 2023-12-16:
Whoa I am really suprised that this was already a topic at Analogue. That makes me sad and happy at the same time.

But what do you mean with "custom cartrigde"? Like FXPak Pro for SNES? I was hoping there is a "software only solution" and using the USB connection of the Docking station.....

So if you are motivated sometime, I would be a guaranteed user :p

agg23, 2023-12-16:
The USB port on the Pocket is not user accessible. I suppose theoretically Analogue could add native support for this, but I doubt it would happen. The custom cart would simply be to provide USB IO to the Pocket.

Quaschynock, 2023-12-16:
Okay thanks for the detailed information. Now I understand the construction and what would be necessary to implement autotracking. 

Thanks, I hope it will be possible at some point.