agg23/openfpga-SNES #93: Audio lags and cuts out randomly

itsdemboys, 2024-01-03:
The two biggest are NHL '96 as well as Taz-Mania. The former seems to be having issues loading in voices, and one time I even completely lost audio after scoring my first goal. In Taz-Mania, it's having trouble playing the music consistently, cutting in and out quite a bit throughout each level. Just wondering if anyone else has experienced similar issues with this core, and if they might be able to recommend any fixes?
I'm currently on Analogue OS 2.0. Here's a [video of NHL](https://youtube.com/shorts/u-deL_EnpDM) in action and as you can hear, the music is slowed way down compared to what it should normally sound like.

agg23, 2024-01-03:
That's really weird. That's the first I've heard of something like this, and I don't see any similar mentions for the MiSTer core. Both of those games don't use special chips.

Are you using NTSC versions of both games?

itsdemboys, 2024-01-03:
Yes, NTSC for both. I even scrapped the rom I was using on NHL last night and tried a different one in case mine was bad, and am having the same issue. Do you think I might have more success with a PAL rom?

agg23, 2024-01-03:
No, NTSC would have better support, which is why I asked.

itsdemboys, 2024-01-04:
Interesting development, someone else over on Reddit also having the same issue with the NHL games [[]](https://www.reddit.com/r/AnaloguePocket/comments/18yk5wi/comment/kgcaokm/?context=3)(url)

tramtrist, 2024-01-05:
Its me.. I'm OP from reddit.  Fire up the rom and you'll freeze at some point between the end of the 1st period and middle of the 2nd every time.  Have yet to get through a dozen attempts at playing a full game