agg23/openfpga-SNES #95: Crash ROM

Vashfairytail900, 2024-01-22:
This Hackrom's.
DONUT PLAINS 4 (course #6), STAR WORLD 5 (course #136), FOREST OF ILLUSION 1 (course #11E)
The game freezes when I start the course of
SNES actual machines and other emulators (BSNES,Snes9x etc) are working fine.
You can download it from the following link.
Overworld can be moved freely because of the debug patch.

https://xgf.nu/3ogHu

Vashfairytail900, 2024-01-22:
P.S.: This is Hackrom from Super Mario World.

agg23, 2024-01-22:
Please don't post links to copyrighted content.

What is the name of the hack, and where can I get the patch?

Vashfairytail900, 2024-01-23:
I apologize for this.
From this profile, select "View" under "Files" and the file smwx.ips is the relevant hackrom.

Vashfairytail900, 2024-01-23:
Forgot the link. My apologies.

This Page
https://www.smwcentral.net/?p=profile&id=29537

agg23, 2024-01-23:
I was unable to successfully apply the patch. What's the hash of the intended ROM?

In any case, the issue is almost definitely a problem with the upstream MiSTer core, and can not be addressed here. Beware, however, that they will not want you filing a bug report before first testing on MiSTer.

Vashfairytail900, 2024-01-23:
Try another patch.
The patch is called smwx.bps on the same page.
It should work if you apply it to the original SMW ROM.

agg23, 2024-01-25:
The `bps` works. I see that the game boots. For some reason I didn't expect that the level names you mentioned were vanilla and I would have to play through a significant portion of the game to reach them. I also have no desire to go find a completed save file for normal SMW, if that will even load in your hack.

As I said above, it's going to be a upstream MiSTer core issue, and I'm not going to do anything about it. While there may be something interesting to learn from what is breaking your hack, no one cares to make one random romhack work on the core.

closed 2024-01-25 by agg23