agg23/openfpga-SNES #98: [Feature Request] Cheat code support

flanter21, 2024-02-12:
I was wondering whether cheat code support (gameshark/game genie) is being considered.

agg23, 2024-02-12:
This has been requested many times before across my cores. Without Analogue building a specific UI for cheats, it's very difficult to build a reasonable mechanism for them. I actually have some ideas on how to do it, but it's not something I will be exploring any time soon.

flanter21, 2024-02-12:
Understood. Didn't see any similar issue for this one which is why I asked. I do note that there is cheat support in the upstream mister core if/when you decide to return to it.

agg23, 2024-02-12:
Yes, MiSTer has the required supporting UI though. That's the hard part, because these cores don't have any available space for a CPU to display a cheat UI. MiSTer runs that on the physical (hard) CPU, so it doesn't have that problem.

Thank you for creating the issue. It's useful to track.