agg23/openfpga-SNES #99: Fully buffered and zero delay modes in agg23 Snes core?

roflcopter777, 2024-02-21:
This is probably a silly question, or maybe I'm too sensitive to the dropped frames every ten seconds or so, but I figured I'd ask here. On the Super NT, there are the buffered modes, fully buffered and zero delay, which gets rid of the occasional skipped frame, at the cost of extra latency but, it results in a smooth scrolling. On the Analogue Pocket's OpenFPGA cores, I seem to be very sensitive and notice the skipped frames every ten seconds or so, is there any way to add or force buffered modes that get rid of this issue? I really don't care about the extra input delay, I just don't want to see that random skipped frame as I notice it very easily. Thank you :3

PS: To be honest, I don't have high hopes of this being addressed given its seemingly trivial nature, and that no one else has bothered to report it and/or don't notice it, don't want to be cynical, but I've had bad luck reporting these issues in the past.

agg23, 2024-02-21:
Are you referring to dropped frames when docked due to the resampling from 59.97Hz to 60.00Hz? I hadn't considered this; I'm not actually sure how this is generally handled (and why it couldn't be automated by the Pocket firmware).

roflcopter777, 2024-02-22:
> Are you referring to dropped frames when docked due to the resampling from 59.97Hz to 60.00Hz? I hadn't considered this; I'm not actually sure how this is generally handled (and why it couldn't be automated by the Pocket firmware).

I don't know, honestly, I feel like I'm making a mountain out of a molehill since other people either don't notice it and/or aren't sensitive to this issue. I don't know if it's widespread to all the OpenFPGA cores or not, and I want to apologize if this is a waste of your time, since the issue could be deeply rooted somewhere. I don't even have a way to capture video properly, but yes, that's the issue in a nutshell; 59.97 Hz instead of being rounded to 60.00; I could handle the input lag if it meant smoother video. I may just stick to using my Steam Deck and Bsnes in the mean time. Sorry again for bothering ya. And thanks for the quick reply.

roflcopter777, 2024-02-22:
Yeah, I'll take that as an issue that Analogue will never address.

TheOldCamper, 2024-12-21:
Well this is indeed a very noticeable issue, especially for Snes games, and deal-breaker for the dock... Hope a fix can be made someday.

TheOldCamper, 2025-10-03:
> Are you referring to dropped frames when docked due to the resampling from 59.97Hz to 60.00Hz? I hadn't considered this; I'm not actually sure how this is generally handled (and why it couldn't be automated by the Pocket firmware).

Unfortunately, to date, the issue remains, I should have mentioned that it’s the SNES hardware outputting at 61 Hz and being downsampled by the dock to a 60 Hz HDMI signal. So while the difference is small, it’s not negligible…