agg23/openfpga-SNES #102: Gundam Wing: Endless Duel won't load

shizzmoney, 2024-06-29:
Trying to play "Shin Kidou Senki Gundam W - Endless Duel (Japan)" and it won't load.  Black screen.

fooness, 2026-10-24:
> Trying to play "Shin Kidou Senki Gundam W - Endless Duel (Japan)" and it won't load. Black screen.

The original ROM file, or with the Aeon Genesis applied? The original works for me, the patched version results in a black screen.

Same on MiSTer FPGA.

fooness, 2026-10-25:
@shizzmoney If my previous assumption [you using the Aeon Genesis translation patch] was correct, you might want to have a look at a possible solution to your problem here: https://github.com/MiSTer-devel/SNES_MiSTer/issues/521#issuecomment-6398422646