agg23/openfpga-SNES #108: CPU and SuperFX turbo options show no speedup in VitorVilela7's SnesSpeedTest

nitro2k01, 2024-11-29:
I tried running [VitorVilela7's SnesSpeedTest](https://github.com/VitorVilela7/SnesSpeedTest). I might be misunderstanding how either the turbo options or the test in question work, but my understanding is that the turbo option overclocks either CPU and allows them to run more instructions per second, and that the speed test measures that exact metric. But running the test with the options off or on shows no difference at all in any of the tests. I know that the options do _something_ (ie that the menu action is being forwarded to the core) because they're able to crash some games. Am I misunderstanding the function of the turbo options? Is it modifying the CPU speed in some way that can't be detected by the test?

I've tried this on core version 0.4.0 and 0.4.3 with Analogue OS 2.3.

Both turbo options off:
[image]

Both turbo options on:
[image]

nitro2k01, 2024-11-29:
I also tried PaulB's test, which does show a 2x difference.

[image]

[image]

nitro2k01, 2024-11-29:
Further, someone tried VitorVilela7's test on an actual MiSTer and it didn't do anything there either. Even if this is an expected result, I'm curious _why_ it's expected and how things work under the hood. And considering that the test doesn't show a change on a MiSTer, I might ask on the upstream repo instead.

agg23, 2025-03-08:
MiSTer doesn't really officially support the turbo options. They act weirdly and break games. The documented compatibility (as minimal as it is) is [available here](https://github.com/MiSTer-devel/SNES_MiSTer/blob/master/SNES_Turbo.md). If you want to report problems, you should do so upstream, but ideally don't mention that you're using the Pocket, as they'll just close your issue.